riscvarchive/riscv-v-spec #854: Misleading comments in "7.8.2. Vector Strided Segment Loads and Stores example"

viclabcodasip, 2023-01-27:
Hi, I think there might be a possible errata in the comments in the example in https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#782-vector-strided-segment-loads-and-stores. 
 
Let's note that it is the description what defines the specs and not the example. However, the example can lead to confusion if not clear/correct.
 
My reasoning for which I consider there might be a errata in the comments in the specs example is the following:
 
- Consider the example
- Consider the value in x6 is 1

- For i=0 the value x5+2 is written to v6[0]
- For i=1 the value x5+2 is written to v5[1]
- For i=2 the value x5+2 is written to v4[2]

- For i=1 the value x5+3 is written to v6[1]
- For i=2 the value x5+3 is written to v5[2]
- For i=3 the value x5+3 is written to v4[3]

 etc
 
This means that, if the comments in the example were correct, the instruction is writing the same sequence of values (but displaced between each other) in the different destiny vectors. 
 
This is not doing what (by my understanding) the description does:

"Vector strided segment loads and stores move contiguous segments where each segment is separated by the byte-stride offset given in the rs2 GPR argument."

If the comments were true, the segments are overlapping instead of being separated by the given offset, as the description says (instead of rs2=1, overlapping would mean the value is rs2=-1).
 
Additionally, a clearer or more detailed description of the instruction can help.

aswaterman, 2023-01-27:
The stride specifies the distance between the start of a segment and the
start of the next segment, not the distance between the end of a segment
and the start of the next. I’ll let others debate whether or how best to
clarify the text.

viclabcodasip, 2023-01-27:
Thank you very much for the clarification, Andrew,