riscvarchive/riscv-v-spec #859: Question about the instruction of vfrsqrt7.v?

lico614257, 2023-02-07:
The following is a description of the vfrsqrt7.v instruction in riscv-v-spec-1.0 :

More precisely, the result is computed as follows. Let the normalized input exponent be equal to the input exponent if the
input is normal, or 0 minus the number of leading zeros in the significand otherwise. If the input is subnormal, the
normalized input significand is given by shifting the input significand left by 1 minus the normalized input exponent,
discarding the leading 1 bit. The output exponent equals floor((3*B - 1 - the normalized input exponent) / 2). The output sign
equals the input sign.

Question: I don't know what value B is in the formula((3*B - 1 - the normalized input exponent) / 2)?

topperc, 2023-02-07:
I think B is the exponent bias as noted at the start of the vfrec7.v table in the next section

```
The following table describes the instruction’s behavior for all classes of floating-point inputs, where B is the exponent bias:
```

lico614257, 2023-02-07:
I think you're right

topperc, 2023-02-09:
Opened #861 to clarify this in the spec.