riscvarchive/riscv-v-spec #863: Does vmv1r.v depend on vtype?

jrahmeh, 2023-02-17:
Should the vmv1r.v/vmv2r.v/vmv4r.v.vmv8r.v raise an illegal instruction exception when vill is set?
In section 16.6 there is a note that states:
```
These instructions are intended to aid compilers to shuffle vector registers without needing to know or change vl or vtype.
```
If theses instructions do not depend on vtype (as implied by "without needing to know vtype"), then they should not be affected by vill being set.  Is that correct?

nick-knight, 2023-02-17:
Correct.

<div dir="auto">
<p dir="auto">If the <code>vill</code> bit is set, then any attempt to execute a vector instruction
that depends upon <code>vtype</code> will raise an illegal-instruction exception.</p>
</div>
<div dir="auto">

Note | vset{i}vl{i} and whole-register loads, stores, and moves do not depend upon vtype.
-- | --

</div>

</div>

https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#344-vector-type-illegal-vill

closed 2023-03-13 by kasanovic

aswaterman, 2023-04-03:
However, the non-normative note that @nick-knight referenced is incorrect, as it contradicts the normative text that defines `vmv<nr>r.v`, which says the instructions _do_ depend on `vtype`.  I've created a PR to delete the offending non-normative text.  https://github.com/riscv/riscv-v-spec/pull/872