riscvarchive/riscv-v-spec #866: Question about vstart?

jinyinghan324, 2023-03-13:
Our implementation will never take interrupts during any vector instruciotn, including vector reduction, permutation, mask instructions. Is is permitted to raise an illegal instruction exception when attemptin to exectue any vector instruction when vstart is nonzero?

aka2en, 2023-03-13:
Yes, it is permitted. Check spec:
```
Implementations are permitted to raise illegal instruction exceptions when attempting to execute a vector instruction with a value of vstart that the implementation can never produce when executing that same instruction with the same vtype setting.
```

huangxiansheng123, 2023-03-13:
[image]
Does it also include the vmv.x.s instruction? @aka2en

nick-knight, 2023-03-13:
Yes: according to my reading of the spec, there are no exceptions to the provision that @aka2en quoted.

closed 2023-03-13 by kasanovic