riscvarchive/riscv-v-spec #867: Segmented vloads with offset vector register EMUL*NF > 8 seems ambiguous

amorenosmd, 2023-03-14:
The spec seems to be ambiguous with respect to illegal indexed segmented vector loads. In particular, consider the following sequence:

```
vsetvli a1, t0, e8, ta, ma
vluxseg6ei64.v v8, (x5), v16
```

The data vector register group EMUL is 1, while the offset vector register group has EMUL=(64/8)*1 = 8. The spec says the following about segmented vector loads:

> The EMUL setting must be such that EMUL * NFIELDS ≤ 8, otherwise the instruction encoding is reserved.

Interpreting this directly, I would assume that the previous sequence of instructions is illegal, since for the offset vector register group EMUL * NFIELDS = 8 * 6 > 8. But reading a bit more into the spec, one could argue that this should be legal because we are not accessing more than one vector register group of 8 elements. In fact, the next line after the previous quote says:

> The product ceil(EMUL) * NFIELDS represents the number of underlying vector registers that will be touched by a segmented load or store instruction. This constraint makes this total no larger than 1/4 of the architectural register file, and the same as for regular operations with EMUL=8.

Stating that the goal of such restriction is to avoid accessing more than one 8-register vector group, which we are not doing in my example.

This feels a bit ambiguous to the point where I'm not quite sure which of the two is the correct interpretation. Can someone confirm which of the two interpretations is the correct one? Would it be possible to add some clarification in the spec?

nick-knight, 2023-03-14:
I agree that the wording "the EMUL setting" is confusing, because EMUL is a property of a vector operand to an instruction, and different operands can have different EMULs. Like you point out, below in [section 7.3.8](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#783-vector-indexed-segment-loads-and-stores) it says:

> The data vector register group has EEW=SEW, EMUL=LMUL, while the index vector register group has EEW encoded in the instruction with EMUL=(EEW/SEW)*LMUL.

I've always read the spec as mandating EMUL * NFIELDS <= 8 only for the "data" vector operand. But I guess you could view it as also applying to the "index" vector operand, with NFIELDS = 1.

nick-knight, 2023-03-14:
Here is my proposal: #868

aswaterman, 2023-03-14:
Merged, thanks for helping to improve the spec.

PR #868 (2023-03-14): Clarify EMUL constraint for indexed segment ops
Closes #867 .

Another option is to make this clarification earlier. I don't have strong feelings. Someone else is welcome to propose that.