riscvarchive/riscv-v-spec #873: Largest element index is vague

jrahmeh, 2023-04-03:
Section 3.7 says: 
```
The use of vstart values greater than the largest element index for the current SEW setting is reserved.
```
Is the largest element index computed using the current group multiplier, or the largest group multiplier (8)?
For load and store operations with explicit element width (example: vle8.v), do we use the current SEW in vtype or the effective element width (8 for vle8.v)?

hdelassus, 2023-04-18:
This was fixed by #850 (emphasis mine):
> The use of vstart values greater than the largest element index for the current *vtype* setting is reserved.

I think it would make sense to me if the EEW applied to the largest element index calculation for load/store ops with explicit element width.

aswaterman, 2023-04-18:
@hdelassus Yes, you're definitely right about that.  Can you make a PR that makes the wording more precise?  I think it needs to factor in not just EEW but EMUL, right?

aswaterman, 2023-04-18:
Actually, on further thought, isn't it correct as-is?  Because the larger-EEW operands also have EMUL adjusted by the same amount, simply using SEW and LMUL (as the current text suggests) is equivalent to using EEW and EMUL for the widest type.

jrahmeh, 2023-04-18:
How about: "The use of vstart values greater than the largest element index is reserved.  The largest element index is determined using the effective element width and effective group multiplier of the current instruction."

hdelassus, 2023-04-18:
> @hdelassus Yes, you're definitely right about that. Can you make a PR that makes the wording more precise? I think it needs to factor in not just EEW but also EMUL, right?

Sounds about right to me. Also, @jrahmeh's suggestion LGTM.

> Whole-register loads and stores are a counterexample to the struckthrough text, since their EEW:EMUL ratio might not be equal to the SEW:LMUL ratio. 

Also mask loads and stores, which have EEW=8 and EMUL=1, right?

aswaterman, 2023-04-18:
> Also mask loads and stores

Yes, those, too.

Would either of you mind making a PR?

nick-knight, 2023-04-18:
Not 100% sure, but we might need a similar modification to a subsequent statement:

> Implementations are permitted to raise illegal instruction exceptions when attempting to execute a vector instruction with a value of `vstart` that the implementation can never produce when executing that same instruction with the same `vtype` setting.

I've always been a little confused about potential overlap between this statement and the earlier one being debated in this issue. So take my comment with a grain of salt.

aswaterman, 2023-04-18:
@nick-knight I think that one should stay as-is.  Implicitly, it admits the possibility that the constraints on vstart depend on the opcode, or on vtype, or on a combination of the opcode and vtype.  So it covers all bases.