riscvarchive/riscv-v-spec #879: Why would you mask register after 32 bit addition? Help needed

playXE, 2023-05-05:
B3 JIT in WebKit emits the following instruction sequence for 32 bit operations:
```riscv
    0000004002a97038: <0016061b> addiw x12, x12, 1
    0000004002a9703c: <02061613> slli x12, x12, 32
    0000004002a97040: <02065613> srli x12, x12, 32
```

Why is this done?

nick-knight, 2023-05-05:
This repo is for the RISC-V vector extension. I don't see anything related to vectors in your code snippet.

It sounds like your question is about compiler-generated code. Perhaps that compiler's decelopers would be best equipped to explain their compiler's behavior.

playXE, 2023-05-05:
Oh sorry, I mistaken the repo

> This repo is for the RISC-V vector extension. I don't see anything related to vectors in your code snippet.
> 
> It sounds like your question is about compiler-generated code. Perhaps that compiler's decelopers would be best equipped to explain their compiler's behavior.

closed 2023-05-05 by playXE