riscvarchive/riscv-v-spec #881: How many registers are used for "vector load/store" when EMUL and LMUL not equals

miloee, 2023-05-19:
Let me take an example:
Firstly, set vector settings to "VLEN=128 SEW=32 LMUL=8".
Use "vle16.v" to load.
The EEW of "vle16.v" is 16, which means EMUL=(EEW/SEW)*LMUL=4.

I can image 2 options to implement:
1. Only load 4x128bits to vector register.

topperc, 2023-05-19:
It will load 4x128 bits into 4 vector registers. The register for the load can be any valid LMUL=4 register. v0, v4, v8, v12, v16, v20, v24, v28.

miloee, 2023-06-25:
> It will load 4x128 bits into 4 vector registers. The register for the load can be any valid LMUL=4 register. v0, v4, v8, v12, v16, v20, v24, v28.

Sorry to reopen this issue. 

I would like to know if the EEW = 16 and EMUL = 4. There is no description of EEW and EMUL for unit-stride load/store and stride load/store. I 'd like to clarify EEW and EMUL's configuration for the case.

Furthermore, why do vector load/store instructions have their own EEW configuration? Can you provide an example to illustrate it?

closed 2023-06-25 by miloee