riscvarchive/riscv-v-spec #884: Question about FRM of conversions.

zhongjuzhe, 2023-05-29:
Hi, I wonder these following instructions:

1. vfcvt.f.xu.v vd, vs2, vm/vfcvt.f.x.v  vd, vs2, vm
2. vfwcvt.f.xu.v vd, vs2, vm/vfwcvt.f.x.v  vd, vs2, vm
3. vfwcvt.f.f.v vd, vs2, vm
4. vfncvt.f.xu.w vd, vs2, vm/vfncvt.f.x.w  vd, vs2, vm
5. vfncvt.f.f.w vd, vs2, vm

Do they depend on FRM (floating-point rounding mode register)

Thanks.

aswaterman, 2023-05-29:
In general, widening conversions are unaffected by the rounding mode, but narrowing conversions are affected by the rounding mode.

zhongjuzhe, 2023-05-29:
> 

Thanks. Andrew.
What about same element width:
vfcvt.f.xu.v vd, vs2, vm/vfcvt.f.x.v vd, vs2, vm
This is integer to float conversion, same width, for example, int32 to float.
Is it affected by rounding mode?

aswaterman, 2023-05-30:
FP to integer conversions always depend on the rounding mode, regardless of width.

For integer to FP conversions, the answer depends on the widths.  For example, the FP64 mantissa is wide enough that all int32/uint32 values can be represented exactly.  But not all int32/uint32 values can be represented exactly in FP32.  In general, we need the significand (including the implicit leading 1 bit) to be greater than or equal to the integer width (excluding any sign bit) to avoid rounding.

As a side note, don't forget about this caveat in the spec: "All vector floating-point operations use the dynamic rounding mode in the frm register. Use of the frm field when it contains an invalid rounding mode by any vector floating-point instruction—​even those that do not depend on the rounding mode, or when vl=0, or when vstart ≥ vl--is reserved."

zhongjuzhe, 2023-05-30:
Thanks Andrew. Closing it.

closed 2023-05-30 by zhongjuzhe