riscvarchive/riscv-v-spec #888: Question about setting spike

huikmn, 2023-06-13:
We are using the latest spike: Spike RISC-V ISA Simulator 1.1.1-dev as our reference mode. In our design, we choose all 1s as our agnostic policy for masked and tail elements. But the spike seems choose undisturbed as default, and we haven't find any settings to change this. So out result comparesion is always wrong. Is there any way to change the agnostic policy of spike to all 1s? Thank you!

furuame, 2023-06-14:
The proper place to file this request is https://github.com/riscv-software-src/riscv-isa-sim

huikmn, 2023-06-14:
> The proper place to file this request is https://github.com/riscv-software-src/riscv-isa-sim

ok, thanks, i wll ask again

closed 2023-06-14 by huikmn