riscvarchive/riscv-v-spec #894: Is vmsbc exactly the same as vmsltu with vm=1?

liangkaiwang, 2023-06-23:
Hi all,

I recently start reviewing some instructions, and it looks to me that the functionality of vmsbc can be replaced completely with the vmsltu and vm set to 1. Am I missing something here? 

thanks,
Liang-Kai

nick-knight, 2023-06-24:
This may be the case for the vm=1 forms of vmsbc. But I don't think the vm=0 forms are so readily synthesized. For example, if v1 = v2, and v0 has all bits set, then vmsbc.vvm v3, v1, v2, v0 returns v3 with all bits set. Whereas vmsltu.vv v3, v1, v2, v0.t returns all zeros.

aswaterman, 2023-06-24:
Yeah, I came here to say what Nick said.

Obviously, uarches can take advantage of the similarities between any instructions if they’re so inclined.