riscvarchive/riscv-v-spec #895: question about setting vl

huikmn, 2023-06-28:
According to Vector ISA, when use x0 as both rd and rs for VSETVL instruction, VL value should keep existing value.
[image]
But what if the new VTYPE setting makes new VLMAX < existing vl? In the ISA, this situation shoule be reserved.
[image]
So a exception should be detected? or at least VILL in VTYPE shoule be set?
But we ran this case in SPIKE. Our VLEN is 256.
[image]
It turns out there is no exception raised and VL became new VLMAX. Is there a condradiction between ISA and SPIKE? Which one we should follow? Thanks.

aswaterman, 2023-06-28:
There's no contradiction.  Reserved means the implementation can choose the behavior (with illegal-instruction exception being one of the choices).

huikmn, 2023-06-29:
So you means this reserved situation, every solution is acceptable. Spike chooses one solution doesn't means other ways is wrong. But if we use Spike as our reference mode, we better use the same choice as Spike. Do i undertand it correctly?

aswaterman, 2023-06-29:
I'd actually recommend raising an exception in this case.  For the time being, I suggest manually modifying Spike, since we don't yet have a good general way of configuring these implementation-specific behaviors.

huikmn, 2023-06-29:
OK，Thank you very much!

closed 2023-06-29 by huikmn