riscvarchive/riscv-v-spec #896: Question about min/max floating point reduction and NaNs

WojciechMula, 2023-06-30:
Section "14.3.3. Vector Single-Width Floating-Point Max and Min Reductions" does not state explicitly how NaNs should be propagated. Is it valid to assume that NaNs behavior is the same as FMIN/FMAX -- similarly to floating-point sum reductions? (In SPIKE, the authors assumed no NaN-propagation, which is not in-line with RISC-V spec 2.2.)

aswaterman, 2023-06-30:
It seems the spec didn't spell this out, but the behavior should be the same as for FMIN/FMAX.

Looks like this is wrong in Spike (but Spike's implementation of FMIN/FMAX is correct).

aswaterman, 2023-06-30:
https://github.com/riscv-software-src/riscv-isa-sim/issues/1395

WojciechMula, 2023-06-30:
Thank you for the prompt answer!

nick-knight, 2023-06-30:
As discussed on the Spike repo, I do not see a problem.

nick-knight, 2023-06-30:
Please see #897 for a clarification.

WojciechMula, 2023-07-03:
That's great! Thank you very much