riscvarchive/riscv-v-spec #899: Question about Narrowing Floating-Point/Integer Type-Convert Instructions

bhbruce, 2023-07-10:
Do these instructions support `float to uint16/int16 conversion`?

Section [13.19](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#1319-narrowing-floating-pointinteger-type-convert-instructions) mentions that:
     `A set of conversion instructions is provided to convert wider integer and floating-point datatypes to a type of half the width.`. 

I suppose it supports `float to uint16/int16 conversion`.
However, the example below doesn't mention it.
So I would like to confirm whether these instructions support `float to uint16/int16 conversion`.

furuame, 2023-07-10:
Yes I think so

WojciechMula, 2023-07-10:
They support f32 -> i16 and also f16 -> i8. I don't see any restrictions in the spec.

closed 2023-07-11 by bhbruce