riscvarchive/riscv-v-spec #900: Compares effectively AND in the mask under a mask-undisturbed policy

michael-platzer, 2023-07-11:
[Section 11.8](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#118-vector-integer-compare-instructions) of the spec states:
> Compares effectively AND in the mask under a mask-undisturbed policy e.g,
>
>     # (a < b) && (b < c) in two instructions when mask-undisturbed
>     vmslt.vv    v0, va, vb        # All body elements written
>     vmslt.vv    v0, vb, vc, v0.t  # Only update at set mask

Is this an _observation_ that happens to be true when `vd == v0` or is this part of the _specification_ and takes precedence over the default handling of inactive elements as specified in [section 3.4.3](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#343-vector-tail-agnostic-and-vector-mask-agnostic-vta-and-vma)?

As far as I understand, mask destination tail elements are always treated as tail-agnostic but mask destination inactive (i.e., masked-off) elements must be treated according to the current vector mask policy, which means that inactive elements must not be overwritten in the destination vector register if that policy is set to undisturbed.

In the quoted example the destination vector register is the same as the mask source register. Therefore, an AND between the mask source register and the actual result of the compare has the same effect as if the regular mask undisturbed policy had been applied (all elements in v0 that were 0 before the second instruction stay 0). However, if the destination vector register is not `v0`, then this does no longer apply since there could be any value stored in the inactive elements of the destination vector register.

michael-platzer, 2023-07-11:
The observation is not generally valid, it only applies when the destination vector register is `v0`. This is confusing and should be clarified IMHO.

aswaterman, 2023-07-11:
I don’t find it confusing, but file a PR if you want

closed 2023-07-11 by aswaterman

nick-knight, 2023-07-11:
Could this text simply be deleted? I would support that.

commit fbe5761d (2023-07-11): Clarify that compares only AND in the mask if vd==v0

Fixes #900