riscvarchive/riscv-v-spec #902: What happens if VLMAX ends up less than 1?

benjaminou4412, 2023-07-11:
For example (with VLEN=64), a vset{i}vl{i} instruction could set LMUL=1/8, SEW=32, so that an element can't "fit" in the now 8-bit register group (VLMAX=1/4). Or we could have a more sensible setting of LMUL=1/8, SEW=8, but then attempt to execute a vle16 instruction.

As far as I could find in the spec, the only thing close to explicitly addressing this case is section 6.3, which states vset{i}vl{i} instructions should set vl such that it's less than or equal to VLMAX - so for fractional VLMAX, this necessarily implies vl=0, since vl is an integer. The destination register would then be handled like any other tail elements according to vta. I also found [this thread](https://github.com/riscv/riscv-v-spec/issues/782) stating VLMAX<1 is "reserved", but I'm not sure what that means for us trying to implement something - should we have vset{i}vl{i} set vill, then?

WojciechMula, 2023-07-20:
The spec says: "An attempt to set an unsupported SEW and LMUL configuration sets the `vill` bit in `vtype`". Section 3.4.2. Vector Register Grouping

benjaminou4412, 2023-07-20:
We weren't sure whether "supported SEW and LMUL configuration" refers to SEW/LMUL values that are out of bounds, or specific combinations of SEW and LMUL that result in VLMAX<1. The latter would lead to significantly more complicated implementation logic, as every possible combination of SEW and LMUL would have to be checked for illegality.