riscvarchive/riscv-v-spec #904: When the value of vstart written through the csr instruction is greater than the vlmax under the current configuration, whether the hardware can be treated as an illegal instruction to implement

Wei-VV1995, 2023-07-17:
“The use of vstart values greater than the largest element index for the current SEW setting is reserved”
When the value of vstart written through the csr instruction is greater than the vlmax under the current configuration, whether the hardware can be treated as an illegal instruction to implement？

aswaterman, 2023-07-17:
Yeah, in general, reserved behaviors can be implemented as an illegal-instruction exception.

Wei-VV1995, 2023-07-18:
Thank you very much for your answer !

Wei-VV1995, 2023-07-31:
> Yeah, in general, reserved behaviors can be implemented as an illegal-instruction exception.
If vtype=1, is vlamx equal to 0? If vlamx=0, then any value configured for vstart is treated as an illegal instruction, because in this case, any vstart value is greater than the element index.

Wei-VV1995, 2023-07-31:
If vtype=1, is vlamx equal to 0? If vlamx=0, then any value configured for vstart is treated as an illegal instruction, because in this case, any vstart value is greater than the element index.

aswaterman, 2023-08-01:
Duplicate of #907

closed 2023-08-01 by aswaterman