riscvarchive/riscv-v-spec #906: [question] Does V extension implies F and D?

WojciechMula, 2023-07-20:
I cannot locate any mention of it, but Spike code suggests these two are mandatory; `riscv/isa_parser.cc`:

```cpp
    switch (*p) {
      case 'p': extension_table[EXT_ZBPBO] = true;
                extension_table[EXT_ZPN] = true;
                extension_table[EXT_ZPSFOPERAND] = true;
                extension_table[EXT_ZMMUL] = true; break;
      case 'v': // even rv32iv implies double float
      case 'q': extension_table['D'] = true;
                // Fall through
      case 'd': extension_table['F'] = true;
    }
```

asb, 2023-07-20:
Yes, V implies F and D. V requires Zve64d, and that in turn requires D.

topperc, 2023-07-20:
This is documented in "18. Standard Vector Extensions" section of the vector spec.

```
The V extension depends upon the F and D extensions...
```

WojciechMula, 2023-07-20:
@topperc I overlooked that... sorry for the buzz.