riscvarchive/riscv-v-spec #914: benchmarks for RVV

wanghuibin0, 2023-08-10:
I would like to evalute the V vector extension with other vector ISA, such as ARM SVE, or other SIMD architectures.
Is there any representative benchmark for this task?

camel-cdr, 2023-08-21:
@wanghuibin0 I don't think you'll be able to find something like that until rvv 1.0 CPUs are widely available.

This isn't quite what you are looking for, but I've created a few benchmarks to evaluate how to write performant rvv code: https://camel-cdr.github.io/rvv-bench-results/index.html

I've got results from the only two currently generally available CPUs with rvv support (C906 and C920), but note that they implement the pre ratification version 0.7.1 of the extension.

aitesam961, 2023-10-26:
Here is one but not quite there yet
https://github.com/RALC88/riscv-vectorized-benchmark-suite
and here is what they say "Current implementation is targeting RISC-V Architectures; however, it can be easily ported to any Vector/SIMD ISA thanks to a wrapper library which we developed to map vector intrinsics and math functions to the target architecture."
Hope it helps