riscvarchive/riscv-v-spec #924: vstart and precise abort for chaining case

jeremyccl, 2023-10-17:
hi,
I have some questions about vstart and precise exception.
a chaining case in an in-order implementation, vector load (lmul=4) follows by vector add (lmul=4)
if i would like to chain these two instructions together, 

  (empty)  | vload m4 | vload m3 | vload m2 | vload m1

vadd m4 | vadd m3  | vadd m2  | vadd m1 

1. 
what should the vstart value be on the cycle of vload m4 and vadd m3?
should be the oldest instruction's current element? (but vstart should be reset after vload retired)
or should I implement multiple vstart csr for different vector instruction  (but no compatible to RVV?)

2.
what if vload m3 generate exception? in section 17.1 (no instructions newer than the trapping vector instruction have altered architectural state)
Does this mean no vector chaining when there is a vector instruction which could generate exception in the middle of execution?

aswaterman, 2023-10-17:
For precise vector exceptions, chaining must not be architecturally visible.  It doesn't mean that you can't chain; it just means that you need to detect the possible exceptions before beginning chaining.

jeremyccl, 2023-10-19:
Hi Andrew

Thanks for answering, new instructions need to wait for older instruction trap clean than execution.

How about Question 1, when multiple vector instructions executing parallel (or chain), how to maintain vstart?
Any recommendation?

aswaterman, 2023-10-19:
The answers to these questions go together. The vstart value corresponds to the instruction that took the trap. (All younger instructions must appear not to have begun execution.)