riscvarchive/riscv-v-spec #925: Question about vsrl instruction in RVV spec

LWenH, 2023-10-31:
Hi, RVV maintainers. I think I have a question about the **lg2(sew)** restriction for the **vsrl** instruction here.

For the arithmetic right shift instruction, like "vsra.vv", it's very make sense to restrict the maximum shift amount as **lg2(sew)** bits to
determine the final value to be shifted. 

But I think this restriction may be still not general enough for logical rigth shift, if we got an unsigned value like 128(e8), the maximum value to be shifted will be 7 under the **lg2(sew)** restriction, but if we have a C/C++ program sample like this:
```c
// a, b, c is unsigned
for(int i=0;i<n;i++) {
      a[i] = 128;
      b[i] = 8;
      c[i] = a[i] >> b[i];
}
```` 

the compiler need to zero extend the a, b, c and followed by a truncate instruction like **vnsrl c[i],0** instruction to perform the C/C++ program to get the correct value 0 here.  Under the current vsrl design, I think we can't utilize "vsrl.vv" instruction to perform such code, because the maximum shift amount will be 7 here, which means the result for the c[i] is 1 if we use vsrl.vv here.

It's there any other design considerations for logical right shift to add **lg2(sew)** restriction here? Thank you for your time to read this question.

nick-knight, 2023-10-31:
`unsigned` is equivalent to `uint32_t` on RISC-V (assuming standard ABIs). So this code should be vectorized with SEW = 32, and lg2(32) = 5, and 8 is representable in 5 bits. So I don't see the issue.

LWenH, 2023-10-31:
Yeach, maybe I didn't explain it clearly enough before，the data type should be uint8_t or uint16_t here.

 For uint8_t, the gcc implementation is 1, but for clang/llvm, the reslut would be 0 for this case.

nick-knight, 2023-10-31:
Due to C integer promotion rules, the operands will be promoted to `int`, which is `int32_t` on RISC-V.

Not sure if it's relevant to your question, but note that for lhs >> rhs,
> The behavior is undefined if rhs is negative or is greater or equal the number of bits in the promoted lhs

LWenH, 2023-10-31:
> he behavior is undefined if rhs is negative or is greater or equal the number of bits in the promoted lh

So for the logical right shift instruction, in the case like unsign int (100000.....0  32 bits)  >> 32 ,  is that mean the result would be undefined behaviour?

topperc, 2023-10-31:
> > he behavior is undefined if rhs is negative or is greater or equal the number of bits in the promoted lh
> 
> So for the logical right shift instruction, in the case like unsigned int (100000.....0 32 bits) >> 32 , is that mean the result would be undefined behaviour?

Yes.

closed 2023-10-31 by LWenH

topperc, 2023-10-31:
That doesn't help your problem though.

The shift here is 32 bits due to C integer promotion rules. So shift amounts 0-31 are valid. So your example with 128 and 8 is valid and you want to be able to implement that without vector zero extend and vector truncate.

The vector shift instructions on X86 produce 0 for any shift amount larger than the element size. I think NEON does too. Maybe that behavior for RVV would have helped? But its really too late to change RVV at this point.

You should file a bug against gcc though if they get this code wrong.

topperc, 2023-10-31:
Also here's @LWenH 's rejected patch to LLVM for additional context https://github.com/llvm/llvm-project/pull/69092

nick-knight, 2023-10-31:
In this example, hopefully a compiler could replace the loop with a bzero.

In the case of variable shift amounts, rather than widen and narrow, perhaps it would be more efficient to conditionally merge zeros? (Haven't thought through the details.)

LWenH, 2023-10-31:
Yes, If we could detect such operation or determine the value in both vectors/arrays during the compilation process, transfer into the bzero it did would be a better choice.