riscvarchive/riscv-v-spec #926: Main README dead link

lianakoleva, 2023-11-02:
"GNU toolchain port for v0.8" points to https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-0.8.x, which is no longer a branch of the repo.

aswaterman, 2023-11-02:
If you don't mind making a PR that fixes the broken link, we'll happily merge it.

lianakoleva, 2023-11-02:
I'm not sure what it should be replaced with. All RVV branches have been deleted from that repo. See [riscv-gnu-toolchain/PR1308](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1308) for the discussion.

aswaterman, 2023-11-02:
I don't know how to address this issue if we don't know how to update the README.  Feel free to submit a PR that deletes the offending link.  Closing in the interim.

closed 2023-11-02 by aswaterman