riscvarchive/riscv-v-spec #931: Question about the vl when using element groups.

JerryShih, 2023-12-08:
From the doc:
https://github.com/riscv/riscv-v-spec/blob/e100d4321d7ded558846c32590f9efb0887515da/element_groups.adoc?plain=1#L70

```
The _element_ _group_ _size_ (EGS) is the number of elements in one group, and must be a power-of-two (POT).
```

```
EGSMAX is the largest EGS supported by the implementation.
```
I think the EGSMAX is equal to VLMAX. What's the counterexample for `EGSMAX!=VLMAX`?

```
When AVL > VLMAX, the value of `vl` must be set to either VLMAX or a positive integer multiple of EGSMAX.
```
Assume `EGSMAX==VLMAX`. I think the vl will be VLMAX when `AVL > VLMAX`.

JerryShih, 2023-12-08:
cc @kasanovic

aswaterman, 2023-12-08:
As you quoted, "EGSMAX is the largest EGS supported by the implementation."  The implementation chooses the value of EGSMAX for a given vtype setting, so of course it can be less than VLMAX.

JerryShih, 2023-12-08:
Assume VLEN=128, EGW=128, EGS=4 and EEW=32 for the specific instruction.

Could the implementation have the EGSMAX<16 for LMUL=4 case? Why just make EGSMAX=16?

aswaterman, 2023-12-08:
Consider the case that an implementation doesn't support any instructions that use element groups.  Then obviously it doesn't make sense to have EGSMAX>1, so why have EGSMAX=16?

From an implementation perspective, the rationale is that you normally need to collect all of the elements of a group together to process them.  Supporting very large element groups can therefore become expensive.

Anyway, I think the question is answered.

closed 2023-12-08 by aswaterman