riscvarchive/riscv-v-spec #933: Question about '... traps on ... instructions are always reported with a vstart of 0 ...'

YenHaoChen, 2023-12-25:
Hello,

I am seeking help understanding the vector specification. 

In Section 14, there is a sentence that reads, “Traps on vector reduction instructions are always reported with a vstart of 0.” I have come up with a few possible interpretations but need help deciding which one is intended.
1. The implementation raises an exception trap on vector reduction instructions if vstart is 0.
2. The implementation sets vstart to 0 on raising an exception trap of vector reduction instructions.
3. The implementation reports that vstart is 0 on vector reduction instructions by NOT raising an exception trap.

[image]

Could someone please clarify the intended meaning of this sentence?
Thank you.

YenHaoChen

aswaterman, 2023-12-26:
Option 3 is a logical consequence of the second sentence.  Option 2 is a restatement of the first sentence.  Both are true for different reasons.

Option 1 doesn't follow; it requires a misreading of the first sentence, as if it were rewritten to say "instructions with a vstart of 0".

closed 2023-12-26 by YenHaoChen

aswaterman, 2023-12-26:
You're welcome!  If you think alternative wording would clarify the intent, feel free to suggest it.

YenHaoChen, 2023-12-26:
@aswaterman  Actually, I read the first sentence as Option 2 and feel the sentence works perfectly. The confusion comes when tracing the Spike, which seems not to set vstart to 0 on raising illegal instruction.

[image]

May I provide a PR that sets vstart to 0 for all vector reduction instructions? I would propose to add "P.VU.vstart->write(0);" at the beginning of the macro "VI_CHECK_REDUCTION(is_wide)".

aswaterman, 2023-12-26:
I guess there's an additional implicit, unrelated issue of precise exceptions.  If an instruction doesn't execute, no state should be changed.  So, if `vstart` being nonzero is the reason that an exception is raised, then `vstart` won't be reset to 0.  For these instructions, `vstart` is only zeroed as the result of the instruction being interrupted (which Spike does not model).

YenHaoChen, 2023-12-26:
@aswaterman Understood. The spec allows ```vstart``` to be reset to 0 on raising imprecise exceptions in which the instruction has been executed partially. On the other hand, precise exceptions do not change the state.

I think my confusion started with interpreting ‘always’ as a mandatory requirement. If the spec could use ‘may’ instead, it would clarify the first sentence for me. What do you think of this proposal?

aswaterman, 2023-12-26:
"Traps on vector reduction instructions are always reported with a `vstart` of 0" overrides that constraint.  It holds even with imprecise exceptions.  The reason is that the intermediate state can't, in general, be recorded without destroying other state.  (Consider a widening reduction with `vd`=`vs2`.  If an interrupt occurs after only element 0 has been processed, then writing the intermediate state to `vd` would destroy the unprocessed element 1.)

YenHaoChen, 2023-12-26:
Thank you for helping me understand the detailed vector behavior. 

To the best of my understanding, whether the implementation reports a trap with a ```vstart``` of 0 depends on whether it results in an intermediate state of the vector reduction instruction. If the instruction isn’t interrupted and the state remains the same before the instruction, vstart could retain the same value. The first sentence emphasizes that if a trap interrupts the instruction and the state changes into an intermediate state, the implementation always sets ```vstart``` to 0. Is this correct?

aswaterman, 2023-12-29:
It's really about whether the instruction retires or not.  If a vector instruction retires, it resets `vstart` to 0 as a side effect.  `vstart` isn't reset to 0 if the instruction doesn't retire.

The behavior is something like this:

1. If this vector instruction requires `vstart` be zero, raise an exception (and leave `vstart` intact).
2. If this vector instruction is interrupted, or raises some other kind of exception during execution (e.g. page fault), change `vstart` to point to where the instruction should resume.  (This will only happen for instructions that don't require `vstart` be zero.)
3. If we reach this point, the instruction has retired, so set `vstart` to zero.

YenHaoChen, 2023-12-29:
Thank you for your help. I understand now. Let’s take vector reduction instructions as an example. The complete logical flow is as follows:

The vector reduction instructions require ```vstart``` to be zero; otherwise, the implementation raises an illegal instruction exception. Suppose a trap interrupts an executing vector reduction instruction. In that case, the implementation always reports the trap with zero ```vstart``` to be able to resume the instruction execution after the trap.

aswaterman, 2023-12-29:
Right.  Or, the implementation defers the interrupt until after the instruction retires.