riscvarchive/riscv-v-spec #934: Rounding increment is unclear when d equals to 0

sun-jacobi, 2023-12-28:
In 3.8, we have a pair of functions `roundoff_unsigned` and `roundoff_signed`

```
roundoff_unsigned(v, d) = (unsigned(v) >> d) + r
roundoff_signed(v, d) = (signed(v) >> d) + r
```
and the Rounding increment `r` depends on the rounding mode ( i.e `vxrm`) 

But in the table, `r` is given by expressions like `v[d-1]`. 
But this definition is unclear when d equals 0.

aswaterman, 2023-12-29:
Naturally, shifting by 0 should not result in any rounding.  One fix is to define `v[-1]` to be 0, so that the existing expressions do the right thing.  Another is to change the definition of `r` to be 0 when `d` is 0.

sun-jacobi, 2023-12-29:
> Naturally, shifting by 0 should not result in any rounding. One fix is to define `v[-1]` to be 0, so that the existing expressions do the right thing. Another is to change the definition of `r` to be 0 when `d` is 0.

Thanks for your response, there is actually also a similar case when `vxrm[1:0]` equals `01` and `d` equals `1`
then the r would be `v[0] & (v[-1:0]≠0 | v[1])`

| `vxrm[1:0]` | Abbreviation | Rounding Mode | Rounding increment, `r`|
| ----------- | ----------- | ----------- | ----------- |
| 00 | rnu | round-to-nearest-up (add +0.5 LSB)          | `v[d-1]` | 
| 01 | rne | round-to-nearest-even                       | `v[d-1] & (v[d-2:0]≠0 \| v[d])`|
| 10 | rdn | round-down (truncate)                       | `0` |
| 11 | rod | round-to-odd (OR bits into LSB, aka "jam")  | `!v[d] & v[d-1:0]≠0` |

aswaterman, 2023-12-29:
So we'd also need to say that v[-1:0] is zero by definition, I suppose.