riscvarchive/riscv-v-spec #935: Signed averaging addition through vaadd

sun-jacobi, 2023-12-29:
In 12.2, we introduce `vaadd` for averaging adds of signed integers.

The result would be rounded based on `roundoff_signed` function defined in 3.8. 
```
roundoff_signed(v, d) = (signed(v) >> d) + r
```
The `r` is based on the **rounding mode** (i.e. `vxrm`).

In most programming languages, signed averaging addition is attempting to round the result towards zero. 
AFAIU, This could not be achieved in each kind of rounding mode.

aswaterman, 2023-12-29:
Yes, it's necessary to set `vxrm` to RDN to achieve this result.  Since `vxrm` is caller-saved in the normal ABIs, it's usually cheap to change the rounding mode before a strip-mine loop that wants this behavior.

topperc, 2023-12-29:
I think @sun-jacobi wants 0+-1 to give 0, but RDN would give -1. Using RUP with give 0, but would make 0+1 give 1.

aswaterman, 2023-12-29:
Sorry, I was thinking about the unsigned variant.  Yes, signed RTZ takes quite a few instructions to synthesize.