riscvarchive/riscv-v-spec #937: vmv.s.x/vfmv.s.f behavior question

EventScheduler, 2024-01-15:
We still don't have a precise behavioral definition of vmv.s.x(16.1)/vfmv.s.f(16.2) instructions in the case such as "vstart=1 and vl =2". 
Please refer to "https://github.com/riscv-software-src/riscv-isa-sim/issues/1444".

aswaterman, 2024-01-15:
I wrote the following in the quoted thread:

“This instruction would never set vstart=1, even if interrupted. Therefore, trapping is a legal option (but is not required). I would say it is UNSPECIFIED, so writing vd[0], trapping, or NOP would all be reasonable choices. But this is just an opinion.”

The most important thing is that this case is uninteresting for correct software, since it won’t arise during normal operation. Leaving it as reserved or unspecified won’t harm software, but leaving it unspecified will understandably annoy verification engineers.

Note, however, that trapping must be _allowed_ for backwards compatibility, even if that behavior is left optional. This follows from the clause that implementations may raise an exception when an instruction is executed with a vstart value that instruction could not have produced.

EventScheduler, 2024-01-15:
You're right, I'm a verification engineers. ^_^
Anyway, The designers guys implemented this as NOP, I'll align with them.
Thank you!