riscvarchive/riscv-v-spec #938: Are the vector registers required to have a consistent state at reset?

michael-platzer, 2024-01-24:
Section [3.11](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#state-of-vector-extension-at-reset) states that

> The vector registers can have arbitrary values at reset.

yet,

> The vector extension must have a consistent state at reset.

Does this mean that while the values may be arbitrary, the vector registers still need to be reset to a consistent state (i.e., same arbitrary values on every reset)? Or does "arbitrary values" imply that these values may be inconsistent as well?

jnk0le, 2024-01-24:
It's about not having impossible `vtype`,`vl` config out of reset.
Uninitialization of the working registers is a non issue.

> Does this mean that while the values may be arbitrary, the vector registers still need to be reset to a consistent state (i.e., same arbitrary values on every reset)? Or does "arbitrary values" imply that these values may be inconsistent as well?

Values may be random or hold the contents held prior to the warm reset.

michael-platzer, 2024-01-25:
> It's about not having impossible `vtype`,`vl` config out of reset.

I'm aware of that, this is dealt with in the second sentence of Sect. 3.11. My question is about the implications of the two sentences cited in my first comment.

> Uninitialization of the working registers is a non issue.

The term "vector extension" covers the vector registers as well (see Sect. 3.1), so the requirements of Sect. 3.11 also apply to the vector registers.

> Values may be random or hold the contents held prior to the warm reset.

I do not see any mention of retaining previous contents in the spec. Where do you derive that from?

sorear, 2024-01-25:
"consistent" means "internally non-contradictory", not "the same every reset"; no constraints are applied to the vector registers (all sequences of 32*VLEN bits are valid values of the vector registers), so the requirements imposed by "The vector extension must have a consistent state at reset." on the vector registers are vacuous.

michael-platzer, 2024-02-12:
@sorear Thanks for the clarification. As a non-native-speaker I wrongly assumed that in this context "consistent" meant "staying the same".

closed 2024-02-12 by michael-platzer