riscvarchive/riscv-v-spec #940: The behavior of slideup/slidedown when i is 0

SoufSilence, 2024-01-26:
The spec defines the behavior of slideup/slidedown when i>0, but it does not seem to define the behavior when i=0. Is this an oversight?

> For slideup:
[image]
For slidedown:
[image]

aswaterman, 2024-01-26:
Normally, vstart is 0. The case below the one you highlighted addresses the
0 case. The case you highlighted is relevant only when vstart>0.

SoufSilence, 2024-01-26:
> Normally, vstart is 0. The case below the one you highlighted addresses the 0 case. The case you highlighted is relevant only when vstart>0.
> […](#)

Thank you very much for your quick reply! But please forgive me for not understanding your comment.
Maybe I didn't express myself clearly. What I want to say is that when vstart is 0 and OFFSET is 5, for example, the spec only specifies that vd[1], vd[2], vd[3] and vd[4] should remain unchanged, but the value of vd[0] is not specified.

aswaterman, 2024-01-27:
Ah, it looks like the issue is that we are looking at different versions of the spec.  If you look at the HEAD of master, you'll see that the `<` was changed to `<=` a few years ago.

SoufSilence, 2024-01-27:
> Ah, it looks like the issue is that we are looking at different versions of the spec. If you look at the HEAD of master, you'll see that the `<` was changed to `<=` a few years ago.

Yes we looked at different versions, now I know the way to check the latest version, thanks for your reply.

closed 2024-01-27 by SoufSilence