riscvarchive/riscv-v-spec #943: what's the vd of vcompress in case of vl =0

Kou66666, 2024-03-01:
Hi,

what's the vd of vcompress in case of vl =0? Is old_vd to follow "vstart >= vl", vd unchanged or
follow 16.5, page 87, then vd is all tail, if tail agnostic write "1", then will output all "F"

Thanks,
Kou

16.5, page 87:
The vector mask register specied by vs1 indicates which of the rst vl elements of vector register group vs2 should be
extracted and packed into contiguous elements at the beginning of vector register vd. The remaining elements of vd are
treated as tail elements according to the current tail policy (Section Vector Tail Agnostic and Vector Mask Agnostic vta and
vma).

aswaterman, 2024-03-02:
The general "vstart >= vl" rule applies: "no elements are updated in any destination vector register group".

TogashiHan, 2024-03-05:
> The general "vstart >= vl" rule applies: "no elements are updated in any destination vector register group".
I'd like to double confirm this behavior because I open one similar issue at: https://github.com/riscv/riscv-v-spec/issues/944
,which asked about mask instructions vd result while vl=0, but got different answer.
Should vd be updated with all '1?(if ta update with '1)

In 15.1:
"Mask elements past vl, the tail elements, are **always** updated with a tail-agnostic policy."

aswaterman, 2024-03-05:
The earlier, more general clause takes precedence.

closed 2024-03-05 by aswaterman

TogashiHan, 2024-03-05:
> The earlier, more general clause takes precedence.

Thank you for the clarification.

furuame, 2024-03-05:
> > The earlier, more general clause takes precedence.
> 
> Thank you for the clarification.

@TogashiHan  Oops, sorry for the confusion in #944! Yes in that case, vstart >=vl shouldn't have any side effects on destinations. I think 15.1 rule only applies when vl > vstart.